PAMGuard/PAMGuard #4: Migrating from SVN to Git

alex-konovalov, 2022-01-21:
@douggillespie I've noticed that you've put this on GitHub - this is great! just to add that it's also possible to migrate from SVN to Git and keep all past history of revisions in one repository. Since your SVN repository is public, I don't see a reason why not to do this. Instructions are e.g. at https://git-scm.com/book/en/v2/Git-and-Other-Systems-Migrating-to-Git and we can try this before many things happening here.

douggillespie, 2022-01-21:
Hi Alex,
I’m aware that you can do a full migration. However the repository was a really bad mess on SVN with multiple projects, some of this are obsolete all within the main repository. While it may have been possible to migrate the whole lot, then delete all the unused branches, we decided that leaving the history on SVN and starting a clean repo would be the easier way to go.
Best,
Douglas.

alex-konovalov, 2022-01-21:
@douggillespie ok then! One thing to consider might be to migrate it as a separate repository under the same organisation at https://github.com/PAMGuard/, to keep your old history in one place - may be easier to refer to past commits etc. But perhaps with time that will be needed less and less...

closed 2022-01-30 by douggillespie