PAMGuard/PAMGuard #29: Click classificatio symbols not showing in BT display on PAMGuard open

macster110, 2022-04-21:
When PAMGuard is first opened in viewer mode the click classification symbols do not show properly. For example, if porpoise clicks are red triangles then these are not visible when PAMGuard is first opened - instead the clicks appear as black dots. The symbols are fixed by deselecting and then selecting "Colour by Click Classification" in the right click menu but this is incredibly confusing for users.

Note that is appears this is an issue when opening an older database with a new version of PAMGuard and/or opening the first time after processing data in real time mode. Once fixed, upon reopening PAMGuard is fine. 

Example porpoise data - when first opened the clicks should be red but are black. Note there are two databases. Porpoise_DB opens fine but Porpoise_DB_original, which was generated with an older version of PG opens the clicks with the wrong colour. 

https://drive.google.com/drive/folders/1SCTFgFRcu29QmpvuK3syClh1p-XD7_QF?usp=sharing

douggillespie, 2022-05-30:
The problem with the key will be particularly hard to fix with the new options since the symbol shape and colour are selected based on a series of properties which each click may not have, e.g. a porpoise click may be red because of it's classification as porpoise, but may then get a different colour due to another classifier working on it, or pick up colour from a click train or other grouping of clicks into events. I will try to fix the other problems.

douggillespie, 2024-10-18:
Still thinking about this and have a possible solution where the user can select a single one of the avialable symbol modifiers for a display, and the legend will be based on that one symbol modifier. This doesn't generat ea complete key, but will be better than the current state

lizferguson5, 2025-06-10:
Hi Folks, just thought I would check in on this - I actually have had issues with this for a while - and the re-selection of click classification works for the coloring of clicks in the time bearing/amplitude display, but the legend never updates and all symbols remain black. This isn't the case for older psfxs that I have, it's just whenever creating a newer one. Is it possible to at a minimum have the legend display the colors properly? Even if that requires a re-selection of the "Colour by click classification?"