PAMGuard/PAMGuard #43: UDP Control & Command Syntax

uwgeophys, 2022-07-26:
Hello @douggillespie 

Two issues - I wonder if you can point me the right way. Firstly I appear to be unable to run a UDP command into Pamguard, its compiled OK (thanks for that), the port is open but it disconnects and closes after the first try and only a restart of the software opens it again.

For example:

Pamguard running --port 6666

Another cmd window ---> ncat -u -v localhost 6666 (I am using ncat for sending udp traffic from the nmap package)
command: ping
response: cmd ping not found (with a return/new line after the command in the response)
From then on there are no more responses to any commands & all connections are closed.
After this point the port on pamguard is closed (checked with nmap, and interestingly if you can the port that also crashes it) and it does not reopen till I restart the software. Any clues?

Secondly --- can I please ask for some more clarity on the syntax you expect for the 3 commands listed

1. tellmodule
2. getxmlsettings
3. setxmlsettings

This may be because my attempts at "getxmlsettings" thus far have only resulted in either "No settings" or "Unspecified modules for settings"

For example getxmlsettings "Sound Acquisition" just returns no settings - which seems wrong? 

Part of this is (and i have tried to search but I cannot locate it) there doesn;t appear to be a clear list of available module names, some like Sound Acquisition have a different type of Data Acquisition, others like Array Manager are both the name and type... 

e.g command summary produces:
"<Data Acquisition>Sound Acquisition:ch0,-0.0,-10.4,-46.6,ch1,0.0,-4.6,-22.1,<\Data Acquisition>"

So I presumed the module name is - Sound Acquisition? or is it Data? or is actually just Acquisition (as the kind of root module?)

Finally how did you then envisage someone then parsing the new xml file with setxmlsettings? Something like:
setxmlsettings <module name> <path to xml file> ? or 
setxmlsettings <type name?> <systemType Class=\"java.lang.String\" Value=\"Audio File\"/> (with escape characters used?, just the varrying parts? or the entire file escaped?)

Anyway if you could put an example for each just to read back a default value that would be really helpful.

Thank you very much for your time,

douggillespie, 2022-07-30:
These features are very much NOT a core part of PAMGuard and should be considered as "under development". There is some information on the commands at https://github.com/douggillespie/PAMGuard/wiki/UDP-Commands. You'll have to look in the code to see what the individual modules are returning. As I remember it, the levels from the sound acquisition module will be the mean level, peak level in dB re full scale and rms re full scale.

Regarding the UDP, this is one part I do use regularly so I can't say what's not working. I suggest you run PAMGuard from the debugger and trace through the commands as they arrive to see what's going on. 

the 'getxmlsettings' needs a module type and module name. It will accept wildcards, e.g. 'getxmlsettings "Binary Storage" *' will get settings from the binary storage module.

uwgeophys, 2022-08-01:
@douggillespie thank you for getting back to me. I have had some success in the debugger now its running.

However I appear to not be able to get the system to actually save the settings or reload the new settings.
For example:
```
getxmlsettings "File Input System" *
setxmlsettings <recentFiles Class="java.util.ArrayList" Value="ABC.wav,null,C:\ABC.wav"/> 
```

or 

`setxmlsettings <SETTINGS Class="Acquisition.FileInputParameters" Name="Sound Acquisition" Type="File Input System" Version="1"><recentFiles Class="java.util.ArrayList" Value="C:\Users\Desktop\test\TEST_NAME_HERE.wav,null"/><systemType Class="java.lang.String" Value="Audio File"/><bitDepth Class="int" Value="24"/><skipStartFileTime Class="long" Value="0"/><repeatLoop Class="boolean" Value="false"/><realTime Class="boolean" Value="false"/></SETTINGS>`

I get successfully sent to modules but I can't get it to reload. Is there a way to cause an XML refresh, as simply calling getxmlsettings just replies with the original status.

I appreciate its all bleeding edge, but this really seemed the best to programically set file 1,2,3...X.

macster110, 2024-08-27:
Closing  because of inactivity and will assume resolved.

closed 2024-08-27 by macster110