PAMGuard/PAMGuard #91: Non chronological order processing wav files.

macster110, 2022-12-07:
The behaviour of the Files and Folders Acquisition behaves very differently depending on weather I select just the folder name from the dialog, or whether I navigate into the folder and select all the files within within it. If I select all the files within the folder, then the behaviour seems to be consistent with prior versions of Pamguard, and the files are sorted by time/alphabetically and analysed in chronological order by the timestamp within the file name. If I select the just the folder, then the data appears to be analysed in random (non-chronological nor alphabetical) order. This random order happens regardless of whether I have checked the “Merge continuous files” checkbox or not. For my datasets, I’m not treating the files as continuous anyway – since for my instruments the filename timestamps use the more precise clock than the one in the A/D converter. But for someone who wants contiguous files merged this would likely be a problematic bug.

BrianSMiller, 2024-03-17:
I can confirm that this issue still appears to be present in release 2.02.10, though the order of processing seems less random.  

Also, I suspect that non-chronological processing is more than just an annoyance, and is effectively causing a memory leak. 

For me it seems that datasets with more than a few weeks of audio (which is all my datasets these days) will now run-out-of-memory crash after only a few weeks of data have been processed.  

I have a sneaking suspicion that this might be due to Pamguard expecting data to be chronological and not being able to correctly free the memory of "old data" (i.e. when switching from a more recent to an older file).

BrianSMiller, 2024-03-19:
Apologies Jamie. Please disregard my previous post about 2.02.10 processing files out of chronological order. It seems I was too hasty in my assessment. On closer inspection all evidence indicates that folders of wav files are now being processed chronologically. This was always an issue with wav files for me, and I haven't checked with sud files at all. 

But bottom line is that I'm no longer able to reproduce the issue with non-chronological processing of wav files in 2.02.10, so happy for it to be marked as closed.

closed 2024-08-27 by macster110