PAMGuard/PAMGuard #108: "2-second glitch" in Whistle and Moan Detector

wtleblond, 2023-07-11:
The Whistle and Moan Detector has this issue where false detections (as in contours are drawn when there isn't anything there, including in dead silence) are often produced within the first two seconds or so of sound files. This isn't a recent issue - it was happening as far back as 2020 when I started using PAMGuard. My team uses the WMD extensively, so a good chunk of our data is false detections produced by the 2-second glitch (as we like to call it). This also results in the data map in Viewer Mode having an absolute snowstorm of false markings on it, which makes it impossible to pick out minor encounters from outright false ones.

If this isn't an easy issue to outright fix, a good temporary solution could be to add an option in the WMD to not produce contours within the first two seconds of each file. That would be super helpful for us in the meantime.

Cheers.

douggillespie, 2024-09-20:
Which versions of PAMGuard were you using ? I think this is resolved in the latest. It was caused when telling the sound acquisition to skip x (ie 2) seconds at the start of the file which caused confusion as to the true data start time.

htleblond, 2024-09-20:
Hey, Doug. I was using version 2.02.03, but I just downloaded 2.02.13 and the bug is still there if you keep "skip initial seconds" in Sound Acquisition at zero. However, it looks like skipping the first couple seconds fixes the issue now, which it didn't seem to do the trick before, so that's great.

[image]

douggillespie, 2024-09-22:
Hi Holly,
I'd expect it to still be wrong if you opened old binary data even with a newer version.
What was happening, was that with the old system, it simply skipped the first 2s of the file, so when times were assigned to binary data, which are calculated from the sample number, they were 2s out. This can only be fixed by reprocessing.
To avoid the time offset, the new system doesn't skip anything, but zeros the first seconds (if that options is selected). This keeps the timing simple, but does have other issues.
I hope this clears things up and that you can proceed without having to totally reprocess everything.
If I'm wrong, and you're getting the problem for what you think are other reasons, please let me know.
Douglas

MailScanner has detected definite fraud in the website at "github.com". Do not trust this website: image.png (view on web)<https://github.com/user-attachments/assets/23fbe515-2304-4771-ada1-f4c03cba408f>

-

htleblond, 2024-09-23:
Hi, Doug. The pic I sent was from regular processing mode, not viewer mode, so the false contours were just created on the fly in the new version, not from old binary data.

It's not the end of the world or anything, especially now that we can just skip the first two seconds of each file without it creating false contours (and my plugins have built-in ways of getting around the 2-second glitch anyway), but I figured I'd put that pic up there as an example in case you wanted to try fixing what's left of the issue.

Cheers.