PAMGuard/PAMGuard #167: The file counter does not reset to zero in processing progress bar

macster110, 2024-10-08:
The total number of files in the progress bar does not refresh when a new set of files is selected. For example if a user processes a set of 10 files and then moves to a new folder and processes another 20 then the file number will be 30 instead of 20. 

[image]

douggillespie, 2025-01-20:
Not a bug, but how it's always been. It will contrinue from the next file, unless you open the acquisition dialog, which will reset it. Next version will have a couple of options on the start button to either continue from same point, or reset to start.

douggillespie, 2025-03-11:
Some right click options on start button allow either continuing from same place, or restarting

closed 2025-03-11 by douggillespie