PAMGuard/PAMGuard #171: Speed regression in simple workflow (Wav Folder --> Click Detector) in recent versions compared to 2.01.05

BrianSMiller, 2024-10-11:
My simple .psfx consisting of a "Folder of files" sound acquisition and Click Detector (see attached) runs half as fast in the latest version of PAMGuard 2.02.13 (August 2024) compared to 2.01.05 (from 2020). 

Steps to reproduce: 
1) Add sound acquisition and configure for folder of files (my example was 1700 1-hour long wav files at 2 kHz sample rate 16 bits)
2) Add click detector 
3) Add database and binary output (default settings)
4) save .psfx
5) run .psfx on a folder of wav files using version 2.02.13 and record speed.
6) run attached .psfx on a folder of wav files using version 2.01.05 and record speed for comparison.

I realise that this may not necessarily be considered a a bug, but I thought it might be worth mentioning in case it wasn't already known. For small datasets this slowdown is not really a bit deal. But big datasets (e.g. multi-year), this could mean the difference between completing sometime this week vs. sometime next month.

Obvious workaround is to use the old version if the faster speed is important.

[psfx.zip](https://github.com/user-attachments/files/17338011/psfx.zip)

macster110, 2024-10-16:
Hi Brian, thanks for this. Interesting that it is running slower - it shouldn't do and this is certainly a regression. Can you try removing the click displays and seeing whether the same issues occurs?

We have plans to build a test suite for PAMGuard which should catch these issues in future.

douggillespie, 2025-01-08:
Thanks Brian, I'll try to recreate and work out where the bottleneck is. Can I ask if you get the same affect if you put in a very high detection threshold. There was a major change to the file format of binary files at V2.02, If no clicks are being detected, then file changes can make no difference. the other possibility is noise measurement which was added between those versions too, so you could try turning that off in the click detector.

BrianSMiller, 2025-01-10:
I've taken another look at this using 2.02.14 and tried to simplify things a bit. 

More precise measurements indicate the slowdown is more like 50% slower rather than the 2-3 times slower that I originally estimated. Here are the results: 

| version      | multiThreaded | noiseMeasurement | durationProcessed     | realTimeElapsed HH:MM:SS |
| ------------ | ------------- | ----------------- |---------------------- |-----------------------|
| 2.01.05      | false         | false             | 01:00:00              | 00:00:15               |
| 2.01.05      | true          | false             | 01:00:00              | 00:00:14               |
| 2.02.14      | false         | true              | 01:00:00              | 00:00:22               |
| 2.02.14      | true          | true              | 01:00:00              | 00:00:23               |
| 2.02.14      | false         | false             | 01:00:00              | 00:00:22               |

I've attached a zip with all the .psfx, databases, binary files, and matlab code to produce the table: 
[clickDetectorSpeedTests.zip](https://github.com/user-attachments/files/18369249/clickDetectorSpeedTests.zip)

The timing measurements were made by processing a single file to completion, then comparing the 1st and 61st PCTime column of the Sound_Acquisition.

douggillespie, 2025-01-20:
I'm having trouble recreating this. I've taken four 5 minute files with 25 min gaps between them and have processed with the latest version 2.02.15b and with the older 2.01.05 and if anything the newer version is running slightly faster. Results of processing have been analysed with Matlab which takes the analysis times from the file headers and footers. I ran the experiment with the click display showing, which definitely slows it down due to drawing very large numbers of clicks, and with the empty display tab on top, which stops it redrawing the click display. 

Resulsts are as follows:

>> speedanalysis with click detector display on top
Old version
File \Binaryold\20221024\Click_Detector_Click_Detector_Clicks_20221024_000000.pgdx 17003 data time 17.5s, speed 20.5x
File \Binaryold\20221024\Click_Detector_Click_Detector_Clicks_20221024_003000.pgdx 17056 data time 17.3s, speed 20.7x
File \Binaryold\20221024\Click_Detector_Click_Detector_Clicks_20221024_010000.pgdx 17744 data time 19.8s, speed 18.2x
File \Binaryold\20221024\Click_Detector_Click_Detector_Clicks_20221024_013000.pgdx 18061 data time 18.4s, speed 19.5x
New version
File \binarynew\20221024\Click_Detector_Click_Detector_Clicks_20221024_000000.pgdx 17003 data time 10.5s, speed 34.1x
File \binarynew\20221024\Click_Detector_Click_Detector_Clicks_20221024_003000.pgdx 17056 data time 14.9s, speed 24.2x
File \binarynew\20221024\Click_Detector_Click_Detector_Clicks_20221024_010000.pgdx 17744 data time 29.0s, speed 12.4x
File \binarynew\20221024\Click_Detector_Click_Detector_Clicks_20221024_013000.pgdx 12860 data time 10.6s, speed 24.5x

>> speedanalysis with empty display on top
Old version
File \Binaryold\20221024\Click_Detector_Click_Detector_Clicks_20221024_000000.pgdx 17003 datas, time 5.8s, speed 62.5x
File \Binaryold\20221024\Click_Detector_Click_Detector_Clicks_20221024_003000.pgdx 17056 datas, time 5.0s, speed 72.3x
File \Binaryold\20221024\Click_Detector_Click_Detector_Clicks_20221024_010000.pgdx 17744 datas, time 4.7s, speed 76.0x
File \Binaryold\20221024\Click_Detector_Click_Detector_Clicks_20221024_013000.pgdx 18061 datas, time 5.0s, speed 72.0x
New version
File \binarynew\20221024\Click_Detector_Click_Detector_Clicks_20221024_000000.pgdx 17003 datas, time 4.7s, speed 77.0x
File \binarynew\20221024\Click_Detector_Click_Detector_Clicks_20221024_003000.pgdx 17056 datas, time 4.8s, speed 75.7x
File \binarynew\20221024\Click_Detector_Click_Detector_Clicks_20221024_010000.pgdx 17744 datas, time 4.8s, speed 75.4x
File \binarynew\20221024\Click_Detector_Click_Detector_Clicks_20221024_013000.pgdx 18061 datas, time 4.7s, speed 76.6x

As you can see, the newer version is running faster on average on my computer with my data, though one file did take longer. When not displaying, both are 3 to 4 times faster and are more similar. 

One very significant change was made to the display between these versions. In the old, the display would access the list of click in place (in the PamDataBlock) while it looped through and drew them. This required it to lock that list so that no new clicks could be added while the drawing took place, otherwise Java would throw a ConcurrentModificationException. In the newer versions, The click display copies the list, so has to lock access to the datablock for a much shorter amount of time (it's only copying references to clicks, not the clicks itself) before returning and allowing the click detection algorithm to carry on adding new clicks, which it draws the old ones. 

Which of these two methods is fastest may depend on the data rate. With the test data I'm using, the trigger rate with this configuration was about 6 per second. At a different trigger rate the advantages and disadvantages of the two methods may be different. 

I'd suggest that for crunching a lot of data, you put an empty display on top, so that it's doing less drawing. You can also reduce the time span of the click display to a few seconds - it will still have to loop through all the clicks when it draws, but won't have to calculate so many positions and shapes. 

Also happy to look further if you send a sample of the data you're working with. 
Matlab I use to check the processing time is below.
You're probably familiar with this, but if not, know that the header and footer each contain two dates. The first is called dataDate, which is the time of the data, so is the file start time, and the file start time + it's duration. The analysisDate is the time taken from the computer when the header and footer were written. So in real time these will be identical to dataDate to within a few milliseconds, but for an experiment such as this are the times you processed the data. 
I had two output folders for binary data in sub folders of the root called binarynew and binaryold. There are of course many ways this could be improved to be more informative. 
Functions are in the PAMGuard library on this site at https://github.com/PAMGuard/PAMGuardMatlab

% responding to Brians issue 171
% https://github.com/PAMGuard/PAMGuard/issues/171
% testing with four 5 min wav files and his detector config
root = 'C:\PAMGuardTest\Speed';
dd = dirsub(root, 'Click_Detector_Click_Detector_Clicks*.pgdx');
secsPerDay = 3600*24;
clear et dt nd;
for i = 1:numel(dd)
    [dets, fileInfo] = loadPamguardBinaryFile(dd(i).name);
    dt(i) = (fileInfo.fileFooter.dataDate - fileInfo.fileHeader.dataDate)*secsPerDay;
    et(i) = (fileInfo.fileFooter.analysisDate - fileInfo.fileHeader.analysisDate)*secsPerDay;
    nd(i) = fileInfo.fileFooter.nObjects;
    fprintf('File %s %d datas, time %3.1fs, speed %3.1fx\n', dd(i).name, nd(i), et(i), dt(i)/et(i));
end

Finally, if you're working through a lot of old data, have you checked out the batch processor available in the plugins pages of the pamguard.org website ? This has an option for processing without the GUI.