PAMGuard/PAMGuard #176: MacOS cannot write to NTFS hard drives and so sud file reaidng does not work

macster110, 2024-10-24:
MacOS cannot write to NTFS hard drives. When reading sud files PAMGuard needs to write a file map so it can quickly skip to sections - this cannot be done if the hard drive is read only. 

No good fix to this - Users need a warning message so they know things will not work. Can be easily addressed if users buy a license to use NTFS write/read e.g. from https://shop.paragon-software.com/277/?scope=checkout&cart=263713&cfg=paragon2018

BrianSMiller, 2025-01-01:
Hi Jamie, from my viewpoint, it would be worth thinking about this as a general issue, rather than a specific MacOS issue. 

The actual issue here is that PAMguard is trying to store it's own internal cache of metadata next to the users data. This will fail not just on MacOS NTFS drives, but whenever the user's data is stored in a read only location. The location could be read-only due to technical/legal limitations (MacOS & NTFS) or when the user doesn't have permission to write (read-only drive or user has only read permissions on a network drive). Sometimes this is due to the user's preference (e.g. I often like my wav and pambinary files original and uncontaminated).

In my view, a good fix would be to store PAMGuard's metadata (in this case file map) in a location where PAMGuard has write access. The path of least resistance here would be the location of the PAMGuard log-file since this appears to be always available regardless of whether database and pambinaries are present, though the latter would be fine too, since there's probably no need for a datamap without pambinaries. This change/option would necessitate storing the file maps with unique (but consistent & repeatable) names/locations for each map and PAMGuard keeping track of these (e.g. with a lookup table or equivalent). Sure, this cached metadata won't then travel with the data, but it's better than having to re-create it each time (especially for large datasets).

macster110, 2025-01-05:
Hi Brian - happy new year! Yes, agree with much of this. Having the map files next to the actual files is convenient in many ways but PAMGuard has never required write access to drives with acoustic data and we should not introduce that as a requirement now. 

I would suggest though that the best place might be the Binary file folder. As this module is a requirement for almost all data processing in PAMGuard where you want to save result, then it makes sense to put the files there. The PAMGuard log folder might not be the best solution because.

1. The sudx files can be 10MB each and so over time storage could be an issue. 
2. Usually there is one PAMBinary data folder per instance of PAMGuard which makes figuring out which dataset the sudx files belong to a little easier. It also means it's easy to switch between computers. 

I would also consider saving to the root drive is default and if that fails we save to binary files. Not sure yet whether that's the best idea or not...

Happy to consider counter arguments to all this?