PAMGuard/PAMGuard #184: ROCCA not generating click measurements file

jumoron, 2025-01-02:
I'm using PG viewer to work with big data of clicks. After I label and create an event for the trains, I go to ROCCA measurement, select the events and try to get the result file with the available variables of the manually selected clicks. But I'm not getting any file from it. What am I doing wrong here? Thanks in advance :)

macster110, 2025-01-10:
It is unclear whether this is user error or a bug. GitHub is specifically for bugs and there is guidence [here](https://www.pamguard.org/contact.html) on how to submit bug reports. Please either amend this with the required data or contact PAMGuard support who can put you in touch with ROCCA developers.

jumoron, 2025-01-10:
Thanks for your comment! I amended the report. I'm still not sure if is a bug or user error. I worked with ROCCA a few years ago and I'm following the same steps I previously used, but no csv file is being generated. I will also contact PAMGuard support and ROCCA developers.

macster110, 2025-01-13:
There's defintely an error occurring so this is a bug as it is not returning a warning or being caught by the program. ROCCA is somewhat an external project to PAMGuard and so your best bet for a quick resolution is to get in contact with the ROCCA team. Ask them to update here when they have a solution.

jumoron, 2025-01-13:
Thanks for your comment! I contacted PAMGuard support and I'm waiting for a feedback. I'll ask them to update here once is solved.