PAMGuard/PAMGuard #199: Click Train Detector "CTClassifiersPane: The species codes are not unique" error with more than one Species Classifier

sgkeith, 2025-03-06:
Hi,

I am finding that the Click Train Detector will not run (or even allow its settings tab to be closed) after creating more than one Species Classifier.

First checking that Name and ID under Classifier info are different and are different from any click detector click classification settings, the console spits out below each time you click OK with more than one Species Classifier:
""
CTClassifiersPane: The species codes are not unique
0: 0.0
1: 0.0
ClickTrainAlgorithmPaneFX: Error getting classification settings;
CTClassifiersPane: The species codes are not unique
0: 0.0
1: 0.0
ClickTrainAlgorithmPaneFX: Error getting classification settings;
CTClassifiersPane: The species codes are not unique
0: 0.0
1: 0.0
ClickTrainAlgorithmPaneFX: Error getting classification settings;
""
etc.

Simon

macster110, 2025-03-12:
Thanks for the bug report. Can you please supply a .psfx file if using in real time mode and database and binary file sample if using in viewer mode.

macster110, 2025-06-10:
Going to close this as no reply - please repost if still an issue