PAMGuard/PAMGuard #240: Cannot open JVM dynamic library on MAC

YoussefGha25, 2025-06-10:
Hi Guys,

I installed the latest version (17) on my Mac with the newest version of Mac (Ventura 13.7.6). Unfortunately I get following message during opening software:

Cannot open JVM dynamic library

What can I do?

Thank you in advance.

Youssef

[image]

macster110, 2025-06-10:
Hi Youssef

Ventura is not the latest version -Sequoia is (15.5) which makes me think you have an older Intel based Mac? 

If so, then this will not work as we only support Apple Silicon Macs. If you do have an Apple Silicon Mac then please update to the latest version of macOS or you could try downloading[ Amazon Coretto 21](https://docs.aws.amazon.com/corretto/latest/corretto-21-ug/downloads-list.html) if you don't want to update to see if that solves the issue? 

If that does not work please post here and we will provide more support. 

Cheers!

Jamie

macster110, 2025-06-20:
Closing due to no response

YoussefGha25, 2025-06-23:
Dear Jamie,

Thank you for your help and sorry for the late response. Yes, my Mac is
Intel based so it is not gonna work anyway.

Thank you and kind regards

Youssef